ev3dev/brickstrap #2: Platform names differ for brickstrap system and the brick itself

ddemidov, 2015-01-22:
Machine hardware names differ on  brickstrap:
```
(brickstrap)root@brickstrap:/# uname -m
armv7l
```
and on the brick:
```
demidov@ev3$ uname -m
armv5tejl
```

This results in problems like ddemidov/ev3dev-lang-python#1. Namely, a binary python package created for `armv7l` architecture on the brickstrap system is not accepted by the packaging system on the brick.

Is it possible to sync the platform names on both systems?

The brick runs `ev3dev-jessie-2014-12-01` with the latest updates applied. Same for the brickstrap system.

dlech, 2015-01-22:
Good catch. I am kind of surprised that this has not caused other problems before. It turns out that brickstrap was just using the default settings for `qemu-arm` which is where the `armv7l` came from. I have just pushed a change that sets the proper CPU for `qemu-arm` so that we get the desired `armv5tejl` instead.

ddemidov, 2015-01-22:
Thanks!

That is better, but unfortunately still not perfect. The resulting platform is not `armv5tejl`, but `armv5tel`. I could not find any setting in `qemu -cpu help` that would give the correct result, so I am closing this.

For future reference, explicitly setting the platform name when building the python package (as in ddemidov/ev3dev-lang-python@02b52b4b545657f7beb6d1e76489af8b97ba32be) helped.

closed 2015-01-22 by ddemidov

dlech, 2015-01-22:
I guess I shouldn't have stayed up so late last night. I should have noticed that the `j` was missing. :sleeping: