ev3dev/brickstrap #6: brickstrap qemu errors

BrendanSimon, 2015-10-30:
I'm trying to use brickstrap to create a root filesystem for an ARM A9 embedded system (not ev3dev).

I made a minimial configuration (with no packages) to see if I can build a debian jessie minimal rootfs.

I get as far as configure-packages step when qemu-system-arm throw up errors.  At first qemu couldn't find libpulsecommon-4.0.so.  I managed to get passed that by adding the LD_LIBRARY_PATH below.

Now I get an -E invalid option.  qemu doesn't seem to have a -E option, and I can't work out where this option is being specified.

Any idea how to fix this so I can progress.

Thanks, Brendan.

$ LD_LIBRARY_PATH=/usr/lib/x86_64-linux-gnu/pulseaudio LOG_LEVEL=4 ./brickstrap/brickstrap.sh -b arm-a9-minimal-jessie -d ./brickstrap-builds/arm-a9-minimal-jessie configure-packages
03:26:33 [debug] cmd: configure-packages
03:26:33 [debug] SCRIPT_PATH: /home/brendan/SEPL/brickstrap
03:26:33 [ info] Configuring packages...
03:26:33 [ info] preseed debconf
qemu-system-arm: -E: invalid option

dlech, 2015-10-30:
I have found a fundamental flaw in brickstrap using proot, so I am in the process of wrapping up changes to remove the dependency on proot and use kernel namespaces instead to accomplish the same thing.

I'm actually about ready to push my changes, so stand by for a bit.

BrendanSimon, 2015-10-30:
Sure :)  I guess the problem I am seeing is different though, as many others have been able to build a rootfs from the existing brickstrap.  So any idea where I'm going wrong?

dlech, 2015-10-30:
The changes also alter how qemu is used, so I'm not going to spend time trying to guess until we see the same problem in the new version.

dlech, 2015-10-30:
Alright, I've pushed the changes. Be sure to read the README again. Unfortunately, the image I made does not boot, so apparently, I have broken something, but we can see if you still have the same issue with qemu.

dlech, 2015-10-30:
Oh, and there is a git submodule now, so run `git submodule update --init` after you pull the changes.

dlech, 2015-10-30:
And I just realized I might know what your problem is:

> I made a minimial configuration (with no packages) to see if I can build a debian jessie minimal rootfs.

You have to have packages. The entire rootfs comes from packages. Are any packages installed at all when you run?

dlech, 2015-10-30:
Hmm... I may have made things worse :disappointed: 

The kernel boots, but running `/sbin/init` fails with `-80`

> #define	ELIBBAD		80	/* Accessing a corrupted shared library */

BrendanSimon, 2015-10-30:
Yes, specifying no packages downloads and unpacks a surprising number (94) of deb archives.
I added busybox as a package, and the same thing happens.  Here is a snippet of the output.

    Get:93 http://ftp.au.debian.org/debian/ jessie/main debconf-i18n all 1.5.56 [206 kB]
    Get:94 http://ftp.au.debian.org/debian/ jessie/main gcc-4.8-base armhf 4.8.4-1 [155 kB]
    Get:95 http://ftp.au.debian.org/debian/ jessie/main busybox armhf 1:1.22.0-9+deb8u1 [367 kB]
    Fetched 30.3 MB in 22s (1346 kB/s)
    Download complete and in download only mode
    I: Calculating obsolete packages
    Using directory /home/brendan/SEPL/brickstrap-builds/arm-a9-minimal-jessie/ for unpacking operations
    I: Extracting acl_2.2.52-2_armhf.deb...
    I: Extracting adduser_3.113+nmu3_all.deb...
    -> Processing conffiles for adduser

    <snip>

    I: Extracting util-linux_2.25.2-6_armhf.deb...
    -> Processing conffiles for util-linux
    I: Extracting zlib1g_1%3a1.2.8.dfsg-2+b1_armhf.deb...
    I: Unpacking complete.
    I: Tidying up apt cache and list data.
    I: Tidying up apt cache and list data.

    Multistrap system installed successfully in /home/brendan/SEPL/brickstrap-builds/arm-a9-minimal-jessie/.

    04:26:38 [ info] Copying root files from BOARDDIR definition...
    04:26:38 [debug] BOARDDIR: /home/brendan/SEPL/brickstrap/arm-a9-minimal-jessie
    04:26:38 [debug] ROOTDIR: /home/brendan/SEPL/brickstrap-builds/arm-a9-minimal-jessie
    04:26:38 [ info] Configuring packages...
    04:26:38 [ info] preseed debconf
    qemu-system-arm: -E: invalid option
    brendan@osboxes:~/SEPL$

dlech, 2015-10-30:
Did you define something like `QEMU_COMMAND="qemu-arm -cpu arm926"` in the `config` file?

This has gone away in the changes I just pushed anyway. If you haven't tried the latest changes yet, please do. It uses `qemu-arm-static` now.

BrendanSimon, 2015-10-30:
I'm using `QEMU_COMMAND="qemu-system-arm -cpu cortex-a9"`
I will try using a static version.

BrendanSimon, 2015-10-30:
I can't find any arm static builds of qemu on my Ubuntu VM.  I'm guessing that I need to use another repository with different builds of qemu?

dlech, 2015-10-30:
`sudo apt-get install qemu-user-static`?

dlech, 2015-10-30:
And you need to be using brickstrap from git master, but I literally just push some fixes a few minutes ago. If my SD card reader would cooperate, I would know if I have it sorted or not.

BrendanSimon, 2015-10-30:
Yeah saw that, but the name didn't suggest anything to do with arm or other archs so I assumed it was only for the the host arch.  Looking at the files that were installed I now see qemu-arm-static :)  Brickstrap now completes the process :)

    <snip>
    Setting up dmsetup (2:1.02.90-2.2) ...
    Setting up systemd-sysv (215-17+deb8u2) ...
    Setting up init (1.22) ...
    Processing triggers for libc-bin (2.19-18+deb8u1) ...
    04:50:28 [ info] Running hooks...
    04:50:28 [ info] running hook adduser
    adduser: Only one or two names allowed.
    brendan@osboxes:~/SEPL$

BrendanSimon, 2015-10-30:
I am using Brickstrap from git (just before your changes).  I will try your new stuff soon, but I want to see how far I get with the known Brickstrap first ;-)

BrendanSimon, 2015-11-01:
Am trying your new username based work now.
The README needs to be updated for the `git submodule update --init` command when migrating.

The new namespaces brickstrap seems to work just as good as the proot version for my minimal package selection (actually python3-numpy).  In fact, I'm guessing it works better as it didn't complain about something that the proot version did.

Actually, I don't think either of the scripts succeed for me.  They both stop after running hook adduser.
proot =>

    adduser: Only root may add a user or group to the system.

namespaces =>

    adduser: Only one or two names are allowed.

Further up, both fail initializing the machine id (or right after that).
proot =>

    unable to set CAP_SETFCAP effective capability: Operation not permitted

namespaces =>

    Failed to set capabilities on file '/usr/bin/systemd-detect-virt' (Invalid argument)
    The value of the capability argument is not permitted for a file. Or the file is not a regular (non-symlink) file

BrendanSimon, 2015-11-01:
My bad with hook files.  I copied my hooks scripts from the proot version to the namespaces version and didn't update them for the new method.

dlech, 2015-11-01:
Please open new issues for new problems. It's not clear to me now if you are still getting the capabilities error or not.

Regarding the submodule, I think I am going to just include the code directly rather than trying to maintain a submodule.

dlech, 2015-12-11:
can this issue be closed?

BrendanSimon, 2015-12-12:
yes.  All good since installing `qemu-user-static` package :)

closed 2015-12-12 by dlech