ev3dev/brickstrap #7: Can't run-hooks multiple times

BrendanSimon, 2015-11-03:
The run-hooks target can only be run once, as the adduser hook will fail on subsequent attempts due to the user already existing.

    brendan@osboxes:~/SEPL$ ./brickstrap-namespaces/brickstrap.sh -b sepl-ind-zynq-jessie -d ./brickstrap-namespaces-builds/sepl-ind-zynq-jessie run-hooks
    23:36:48 [ info] Running hooks...
    23:36:48 [ info] running hook adduser
    adduser: The user `robot' already exists.
    brendan@osboxes:~/SEPL$ echo $?
    1

My quick an dirty fix was to append `|| true` to the adduser hook.

    $CHROOTCMD adduser --disabled-password --gecos \"\" robot || true

A better fix might be to delete all users first?  Or delete all users you want to add and leave other existing users untouched?  I guess the main thing is make sure user and group ids are generated as you expect, no matter how many times you execute the run-hooks target.

dlech, 2015-11-04:
By design hooks are only run once and trying to run them again will have unexpected results. Why are you running hooks more than once?

BrendanSimon, 2015-11-04:
Well, to skip the downloading and unpacking of all the debs, when it's been done in a previous step.  If I was updating the adduser hook or modifying my own hooks, I don't want to have to do a brickstrap all every time.  So it's beneficial for this use case to have the scripts be a little smarter, or at least not fail unless there is a real error.

dlech, 2015-11-04:
In that case, I would recommend not using the `all` command. Instead, run each command step-by-step (see `brickstrap --help`). Before your run the `run-hooks` command though, run the `create-tar` command or manually create a copy of your rootfs directory. Then you can use that as a base for creating multiple images.

The `run-hook` and `shell` commands are intended to help in developing a board configuration directory this way.

When you are ready to create a new image, just run the `create-tar` and `create-image` commands.

closed 2015-12-11 by dlech