ev3dev/brickstrap #8: newuidmap error: Could not open proc directory for target (Debian Jessie build host)

BrendanSimon, 2015-11-04:
I have moved to Debian Jessie (because my kernel wont build with Ubuntu Trusty arm gcc-4.8 cross-compiler and will build with arm gcc-4.9 cross-compiler on Jessie)

I didn't get very far.  Any idea how to get user-unshare or newuidmap to work on Debian Jessie.

    brendan@osboxes:~/SEPL$ ./brickstrap-namespace/brickstrap.sh -b bjs-debian-minimal-jessie -d brickstrap-namespace-build/bjs-debian-minimal-jessie all -f
    08:01:55 [ info] Creating multistrap configuration file...
    08:01:55 [ info] running multistrap...
    unshare() failed: Operation not permitted at /home/brendan/SEPL/brickstrap-namespace/user-unshare/user-unshare line 255.
    brendan@osboxes:~/SEPL$ newuidmap: Could not open proc directory for target 2999
    newuidmap failed:  at /home/brendan/SEPL/brickstrap-namespace/user-unshare/user-unshare line 244.

dlech, 2015-11-05:
Actually yes. Read the blog post about user-unshare. There is a setting you have to change that only exists in Debian. We should add this to the brickstrap README.

dlech, 2015-11-05:
FWIW, I'm running the vivid-lts kernel (3.19) on trusty. It's a thing you can do if that help with your gcc issues.

BrendanSimon, 2015-11-05:
OK.  I did the `echo 1 | sudo tee /proc/sys/kernel/unprivileged_userns_clone > /dev/null` trick and it is now doing apt updates :)

Is it worth putting a check for that in the user-unshare script?  Or is it too Debian specific?

I'm not sure running vivid would get me any further as there doesn't seem to be any pre-built cross-compilers for Ubuntu.  According to `https://wiki.ubuntu.com/ToolChain` there are ppa packages here `https://launchpad.net/~ubuntu-toolchain-r/+archive/ubuntu/test` but there only appears to be some gcc-4.8-armhf-cross packages (not 4.9).

There does appear to be a source package for `gcc-4.9-armel-cross` here `https://launchpad.net/ubuntu/+source/gcc-4.9-armel-cross`, so maybe it's possible to build it from source on either Trust or Vivid, but I'm an Ubuntu newbie and am looking for tested pre-built binaries before worrying about build issues.

I'm not sure if the emdebian cross toolchains would work on Ubuntu Trusty or Vivid?  There is also the Linaro prebuilt binaries that could work?

dlech, 2015-11-05:
> Is it worth putting a check for that in the user-unshare script?

Yes, I think it is worth adding, either to the user-unshare script or to the brickstrap script.

> I'm not sure running vivid would get me any further

I was just suggesting using the linux kernel from vivid, not the entire distro (you mentioned you had a kernel issue). But, yes, it looks like `gcc-4.9-arm-linux-gnueabi` is only available in wily.

dlech, 2015-11-05:
> There is also the Linaro prebuilt binaries that could work?

Probably. It has been a while since I looked at their stuff.

BrendanSimon, 2015-11-05:
Good news.  I was able to get brickstrap to create the `ev3-ev3dev-jessie.tar` and `ev3-ev3dev-jessie.img` files on a Debian Jessie build host.  There is something that feels "right" knowing that a Debian Jessie rootfs (armhf) can be generated on a Debian Jessie build host (x86_64).

Now to workout how to combine this with a kernel I built and get it to boot on the Picozed Zynq platform.  Does brickstrap assist with this in some way?  e.g. installing kernel and modules into the filesystem or building initramfs or inittmpfs?  Is there a better forum to discuss this topic?

dlech, 2015-11-05:
I'm on #ev3dev on irc.freenode.net, otherwise this is the best place. Well, actually open a new issue for this particular platform.

dlech, 2015-12-11:
can this issue be closed?

dlech, 2016-01-21:
We've added an error message to brickstrap that explains what to do to fix this, so closing.

closed 2016-01-21 by dlech