ev3dev/brickstrap #9: Modprobe error running brickstrap

woody242, 2015-11-15:
This is my first time attempting to use brickstrap, I was able to use the latest ev3 prerelease on my EV# and now want to start working with ROS on the EV3.  I am attempting to follow your instructions for building an image but get the following error...

modprobe: ERROR: ../libkmod/libkmod.c:557 kmod_search_moddep() could not open moddep file '/lib/modules/3.16.0-30-generic/modules.dep.bin'

Is there a work around for this?

Thanks,
Woody

dlech, 2015-11-15:
When does this error occur? What are you system specs? e.g. are you running Ubuntu trusty?

woody242, 2015-11-15:
Greetings,

I am running trusty, 14.04.2, I have tried in both my working ubuntu  and a completely clean VM.  
I get the same error.

It happens when I run

brickstrap -b ev3-ev3dev-jessie -d ev3dev-ros all

from my working directory ~/work.

Here is the build log

16:06:13 [ info] Copying root files from BOARDDIR definition...
16:06:13 [ info] Configuring packages...
16:06:13 [ info] preseed debconf
16:06:14 [ info] running preinst scripts...
16:06:14 [ info] running /alsa-utils.preinst
16:06:14 [ info] running /apt.preinst
16:06:14 [ info] running /base-passwd.preinst
16:06:14 [ info] running /bash.preinst
16:06:15 [ info] running /bluez.preinst
16:06:15 [ info] running /dash.preinst
Adding 'diversion of /bin/sh to /bin/sh.distrib by dash'
Adding 'diversion of /usr/share/man/man1/sh.1.gz to /usr/share/man/man1/sh.distrib.1.gz by dash'
16:06:16 [ info] running /debconf.preinst
16:06:16 [ info] running /dh-python.preinst
16:06:16 [ info] running /dpkg.preinst
16:06:16 [ info] running /e2fsprogs.preinst
16:06:16 [ info] running /ev3-systemd.preinst
modprobe: ERROR: ../libkmod/libkmod.c:557 kmod_search_moddep() could not open moddep file '/lib/modules/3.16.0-30-generic/modules.dep.bin'

dlech, 2015-11-15:
Ah, `ev3-systemd` is supposed to be blacklisted from running the `preinst` script exactly for this reason. <https://github.com/ev3dev/brickstrap/blob/master/ev3-ev3dev-jessie/preinst.blacklist>.

woody242, 2015-11-15:
That looks much better, indicates it has created file image, a few symlink failures towards the end.  Most seem systemd related so I assume that is expected.

Failed to read /proc/cmdline. Ignoring: No such file or directory
Created symlink from /etc/systemd/system/serial-getty@ttyS0.service to /dev/null.
Failed to read /proc/cmdline. Ignoring: No such file or directory
Created symlink from /etc/systemd/system/serial-getty@ttyS1.service to /dev/null.
Failed to read /proc/cmdline. Ignoring: No such file or directory
Created symlink from /etc/systemd/system/serial-getty@ttyS2.service to /dev/null.
Failed to read /proc/cmdline. Ignoring: No such file or directory
Created symlink from /etc/systemd/system/serial-getty@ttySU0.service to /dev/null.
Failed to read /proc/cmdline. Ignoring: No such file or directory
Created symlink from /etc/systemd/system/serial-getty@ttySU1.service to /dev/null.
Failed to read /proc/cmdline. Ignoring: No such file or directory
Synchronizing state for nfs-common.service with sysvinit using update-rc.d...
Executing /usr/sbin/update-rc.d nfs-common defaults
Executing /usr/sbin/update-rc.d nfs-common disable
insserv: warning: current start runlevel(s) (empty) of script `nfs-common' overrides LSB defaults (S).
insserv: warning: current stop runlevel(s) (0 1 6 S) of script `nfs-common' overrides LSB defaults (0 1 6).
Failed to read /proc/cmdline. Ignoring: No such file or directory
Synchronizing state for rpcbind.service with sysvinit using update-rc.d...
Executing /usr/sbin/update-rc.d rpcbind defaults
Executing /usr/sbin/update-rc.d rpcbind disable
insserv: warning: current start runlevel(s) (empty) of script `rpcbind' overrides LSB defaults (S).
insserv: warning: current stop runlevel(s) (0 1 6 S) of script `rpcbind' overrides LSB defaults (0 1 6).
Failed to read /proc/cmdline. Ignoring: No such file or directory
Synchronizing state for smbd.service with sysvinit using update-rc.d...
Executing /usr/sbin/update-rc.d smbd defaults
Executing /usr/sbin/update-rc.d smbd disable
insserv: warning: current start runlevel(s) (empty) of script `smbd' overrides LSB defaults (2 3 4 5).
insserv: warning: current stop runlevel(s) (0 1 2 3 4 5 6) of script `smbd' overrides LSB defaults (0 1 6).
Failed to read /proc/cmdline. Ignoring: No such file or directory
Created symlink from /etc/systemd/system/samba-ad-dc.service to /dev/null.
16:54:18 [ info] Creating tar of rootfs
16:54:18 [ info] creating tarball /home/woody/work/ev3dev-ros.tar
16:54:18 [ info] Excluding files:

dlech, 2015-11-15:
I don't think the errors about /proc hurt anything, but they have been fixed in git. This log looks normal to me.

moriarty, 2016-01-11:
When I saw this issue mentioning ROS, I tried it and sure enough my ROS instructions were out of date. I left a note in with a link to this issue and @dlech's comment mentioning the ```preinst.blacklist``` file and included a quick wget fix to get around it. 

I removed brickstrap on my VM, and did apt-get upgrade & dist-upgrade, now I have Ubuntu 14.04.3.
Following the brickstrap [install tutorial](http://www.ev3dev.org/docs/tutorials/using-brickstrap-to-cross-compile/), I now have version 0.4.0, and this issue appears to be fixed.

dlech, 2016-01-11:
Thanks for the note. I forgot to close this issue when I released v0.4.0.

closed 2016-01-11 by dlech