ev3dev/brickstrap #13: On Raspberry Pi2: /etc/flash/dtbs does not exist; and kernel DTB is not installed to /boot/flash

cmacq2, 2015-12-25:
The config file `/etc/flash-kernel/db` refers to a file called `boot/flash/bcm2709-rpi-2-b.dtb` but after runnning brickstrap this file is actually located in `/usr/lib/<kernel>/` where `<kernel>` is the Linux kernel package version/name. (There is no copy/symlink in /boot/flash).

Additionally the post-install script of initramfs-tools notices that the directory /etc/flash/dtbs is missing. Is this supposed to link to the `/usr/lib/<kernel>/overlays` directory?

```
update-initramfs: Generating /boot/initrd.img-4.1.15-9-ev3dev-rpi2
Warning: root device /dev/mmcblk0p2 does not exist
flash-kernel: installing version 4.1.15-9-ev3dev-rpi2
Installing new kernel7.img.
Couldn't find /etc/flash-kernel/dtbs/
run-parts: /etc/initramfs/post-update.d//flash-kernel exited with return code 1
dpkg: error processing package initramfs-tools (--configure):
 subprocess installed post-installation script returned error exit status 1
Errors were encountered while processing:
 initramfs-tools
```

cmacq2, 2015-12-25:
Also there's this when installing the kernel:
```
Setting up linux-image-4.1.15-9-ev3dev-rpi2 (1) ...

 Hmm. There is a symbolic link /lib/modules/4.1.15-9-ev3dev-rpi2/build
 However, I can not read it: No such file or directory
 Therefore, I am deleting /lib/modules/4.1.15-9-ev3dev-rpi2/build

 Hmm. The package shipped with a symbolic link /lib/modules/4.1.15-9-ev3dev-rpi2/source
 However, I can not read the target: No such file or directory
 Therefore, I am deleting /lib/modules/4.1.15-9-ev3dev-rpi2/source
```

dlech, 2015-12-25:
> The config file /etc/flash-kernel/db refers to a file called boot/flash/bcm2709-rpi-2-b.dtb but after runnning brickstrap this file is actually located in /usr/lib/<kernel>/ where <kernel> is the Linux kernel package version/name. (There is no copy/symlink in /boot/flash).

`/usr/lib/<kernel>/bcm2709-rpi-2-b.dtb` is the source and `boot/flash/bcm2709-rpi-2-b.dtb` is the destination when `flash-kernel` installs the kernel, so it makes sense that `boot/flash/bcm2709-rpi-2-b.dtb` does not exist if something failed. Not sure why `flash-kernel` is trying to look for `/etc/flash-kernel/dtbs/` though. I'll look into it when I get a chance.

> Also there's this when installing the kernel:

This is a (harmless) bug in the kernel package and has nothing to do with brickstrap.

dlech, 2015-12-26:
I can't reproduce this problem. I just successfully built an image from `rpi2-ev3dev-jessie`. Are you using a modified board configuration directory?

cmacq2, 2015-12-27:
Yes, and I chased it down to a misconfigured `multistrap.conf`. I didn't want to use a default release because that kind of breaks once you go beyond 2 repos or so, and then you really should use apt pinning. Stupid me overlooked that the `/etc/apt/preferences` in `$BOARDDIR/root` is, of course, not used by multistrap without me telling it to! 

This resulted in pulling a newer `flash-kernel` from Debian which *does* use `/etc/flash-kernel/dtbs` so the admin can configure a DTB manually there. (I think it may be broken, because the only way this path can end up in console output is if no DTB is found/the DTB basename is empty -- which the package scripts are supposed to catch and deal with.)

I'm now using an `aptpreferences` key to use apt pinning during multistrap also. (And abusing the fact that I've not installed the ev3dev archive keyring so I get a nice warning message about the ev3dev packages I use because these cannot be authenticated -- this acts as confirmation of whether my pinning works as expected for me.)

dlech, 2015-12-27:
Upstream `flash-kernel` also does not have support for Raspberry Pi at all, so for now, if you want to use `flash-kernel` with RPi, you need the ev3dev version.

cmacq2, 2016-01-15:
This has been resolved for a while now (user error, i.e. my mistake). Time to close the issue.

closed 2016-01-15 by cmacq2