ev3dev/brickstrap #20: Proposal: introduce namespacing to brickstrap.sh

cmacq2, 2016-01-01:
As part of PR #19 the code uses a namespacing convention to avoid name collisions and present a consistent 'guaranteed' execution environment to eval'ed code. For similar reasons such precautions should probably be applied to brickstrap.sh as well.

Proposal: let's apply this namespacing scheme to `brickstrap.sh`, improve on it and document it. This is a low-priority TODO, as it's mostly about improving robustness in corner cases (name collisions), code hygiene and documentation of the execution environment (as exposed to e.g. hooks, `custom-report.sh`).

cmacq2, 2016-01-01:
We should also  document this. Created a starting point [in this branch](https://github.com/cmacq2/brickstrap/tree/document-namespacing/docs)

dlech, 2016-01-01:
:+1: Sound like a good idea to me.

cmacq2, 2016-01-01:
Should I start a PR for the documentation also? 

It's currently still very much WIP, but I think it's worthwhile to merge it anyway because as I see it the namespacing effort is something we should probably introduce piece by piece and then update the documentation as we go. That way the updates to brickstrap.sh are easier to review and also easier to manage w.r.t. impact on the existing configuration trees. Downside is that you will have the documentation/code in a state of flux between the current situation and the namespaced future for a while.

closed 2016-08-13 by dlech