ev3dev/brickstrap #25: Don't export DEBIAN_FRONTEND, DEBCONF_NONINTERACTIVE_SEEN, etc. in brickstrap shell

dlech, 2016-01-11:
We are currently setting...

    export DEBIAN_FRONTEND=noninteractive
    export DEBCONF_NONINTERACTIVE_SEEN=true
    export LC_ALL=C LANGUAGE=C LANG=C
    export PATH=$PATH:/usr/local/sbin:/usr/sbin:/sbin

...globally these are needed for the configure-rootfs and run-hooks steps. It probably doesn't hurt for other steps, but it is not good to have these set for brickstrap shell.

cmacq2, 2016-01-11:
Worth noting: the `PATH` export may also be required for some other parts of brickstrap. We may be using tools from `sbin` paths already. A recent example is: `/usr/sbin/update-binfmts` (though I hardcoded that path). So the path export in particular needs careful review before moving it elsewhere.

cmacq2, 2016-01-15:
Worth noting that in particular `multistrap` is also located in `/usr/sbin`.

dlech, 2016-08-13:
no longer applicable

closed 2016-08-13 by dlech