ev3dev/brickstrap #31: Move documentation to wiki

dlech, 2016-01-20:
We are getting so many features, I'm thinking it would be better to turn on the wiki here rather than trying to cram everything into the readme. Thoughts?

GitHub Pages or ReadTheDocs are possibilities too.

cmacq2, 2016-01-21:
The current brickstrap code already reserves a `docs/` subtree for this purpose (it forbids `-p docs` or equivalent). Instead of a single monolithic readme we should probably have multiple markdown files dedicated to separate topics and then simply cross reference them from the readme.

That also goes someway to the TL;DR effect a long readme might have on new users.

dlech, 2016-01-21:
I was actually thinking the same thing about using markdown in the git tree so that it is readable on GitHub and using something to convert it to man pages that can be included in the debian package. I had a look at `pandoc` and it looks like a good choice for this. FYI, I've moved the project definition (there is only one now) to a `projects/` subfolder, so the check for `docs` on `-p` is probably no longer necessary.

dlech, 2016-08-13:
pandoc is the winner

closed 2016-08-13 by dlech