ev3dev/brickstrap #35: Failes silently when there are command line options after the command

dlech, 2016-01-23:
It used to not matter what order the options or commands are in. Now, if you have an option, e.g. `-f` after the command in the command line, brickstrap fails silently.

cmacq2, 2016-01-23:
The bug is easy to fix: after shifting with `shift $((OPTIND-1))` we need to check that `$# -ge 1` (i.e that at least one more argument remains) or else you have the problem that the last `shift 1` will error out.

This bug is basically in old copy pasted code and presumably it went undetected because it triggers only if you put the command in second to last position.

I'll update PR #29 with a fix.

cmacq2, 2016-01-23:
Since PR #29 has been merged, this issue can be closed now.

closed 2016-01-23 by dlech