ev3dev/brickstrap #51: How to handle postinst scripts that touch /dev/

dlech, 2016-02-03:
I have recently run across a couple packages that try to `chmod` nodes in `/dev/`. Since we have the host system's `/dev` mounted, this fails (as it should). Technically, these should be considered bugs against the debian package. So, the question is what to do about it?

* We could add pre-configure hooks that edit the `postinst` scripts.
* We could just say that you should create a fixed up package instead.
* We could try to just bind mount essential nodes in `/dev` like `/dev/null` instead of mounting the entire directory from the host system.

cmacq2, 2016-02-04:
> We could try to just bind mount essential nodes in /dev like /dev/null instead of mounting the entire directory from the host system.

Makes sense. AFAIK that is what `systemd-nspawn` does also for spawning LXC containers and it makes a good deal of sense to 'match' this whenever feasible because `systemd-nspawn`/`machined` containers probably get a lot more testing/packaging love from upstream Debian than plain, homegrown LXC configs.

cmacq2, 2016-02-04:
>  We could add pre-configure hooks that edit the postinst scripts.

That ... strikes me as very fragile/error prone, not to mention hard to test exhaustively.

dlech, 2016-02-09:
Yeah, it sounds like we should be using system-nspawn instead of user-unshare. It's only available in stretch/wily though and I'm still on trusty. Might be something to consider when xenial comes out.

cmacq2, 2016-02-09:
Problem seems to be that `systemd-nspawn` requires elevated privs though...

dlech, 2016-08-13:
not an issue anymore since we are using docker for bootstraping

closed 2016-08-13 by dlech