ev3dev/brickstrap #55: Absolute rookie: how to compile?

BorisFaz, 2016-03-23:
Hello.
I'm moving my first steps with EV3Dev. I'm not sure but in my opinion the documentation is insuficient at the moment. Well here is my question, is not clear where i have to do things(git download and then compile) in a Linux environment or direclty into ev3 with ev3dev image installed? Is not clear.
I want simply to compile a stupid program for ev3 and execute it, no brickpi. I want to program with C++.
Please help me..

dlech, 2016-03-23:
Brickman is not what you want. Have a look at <https://github.com/ddemidov/ev3dev-lang-cpp>.

BorisFaz, 2016-03-23:
Thanks. The next question is where to do so where is written
### EV3
mkdir build
cd build
cmake .. -DEV3DEV_PLATFORM=EV3
make

And then how to acces to the EV3 brick(there is already the ev3 debian image running)?
I'm trying with putty on windows but  i have problem with the authentication, wich user and pwd?

Thanks and sorry for the questions but i'm a little bit confused on how to start and program it.

dlech, 2016-03-23:
Please open an issue at https://github.com/ev3dev/ev3dev/issues. This is the brickstrap repository and we are not talking about brickstrap here. :smiley:

closed 2016-03-23 by dlech