ev3dev/brickstrap #60: brickstrap tar error

vladshablinsky, 2016-09-20:
Hi, I'm trying to run `brickstrap` on my docker image I imported to docker, but it fails.
I had an image on my OS X machine and exported it to tar into shared folder and then imported it from ubuntu VM into docker. Now I'm getting the following error while running `create-tar`.

I tried both running brickstrap as a shell script and installed using `apt`.

```
➜ viki:git$ ./brickstrap/brickstrap.sh create-tar b01ab378b678 OUTPUT.tar
Checking docker image tar version...
tar 1.27.1-2+b1
Creating OUTPUT.tar from b01ab378b678...
/bin/tar: /brickstrap/_tar-exclude: No such file or directory
/bin/tar: Error is not recoverable: exiting now
```

Here are my tar version and docker on guest.
```
➜ viki:git$ tar --version
tar (GNU tar) 1.27.1
Copyright (C) 2013 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>.
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

Written by John Gilmore and Jay Fenlason.
➜ viki:git$ docker -v
Docker version 1.12.1, build 23cf638
```

Here is my docker on host.

```
➜ vlad:git$ docker --version
Docker version 1.12.1, build 6f9534c
```

What should I do? Thanks!

dlech, 2016-09-20:
You should send a pull request that checks if the file `/brickstrap/_tar-exclude` exists in the docker image. The check should go [here](https://github.com/ev3dev/brickstrap/blob/master/brickstrap.sh#L66) and only set `BRICKSTRAP_TAR_EXCLUDE_OPTION` if the file exists.

The check might look something like this:

    if docker run --rm $BRICKSTRAP_DOCKER_IMAGE_NAME test -f /brickstrap/_tar-exclude; then
        BRICKSTRAP_TAR_EXCLUDE_OPTION="--exclude-from /brickstrap/_tar-exclude"
    fi

dlech, 2016-09-20:
But you probably really want to exclude some files, like `/usr/sbin/policy-rc.d`.

vladshablinsky, 2016-09-20:
@dlech I've opened a PR as you suggested. Btw, It fixes the error, which is not so much of surprise, though. :)

> But you probably really want to exclude some files, like /usr/sbin/policy-rc.d.

 Do I put it into `brickstrap.sh`? Any assistance to regarding the PR is appreciated. I am able to add some additional commits if it's needed.

closed 2016-09-21 by dlech