ev3dev/brickstrap #62: Cannnot open tar file: Permission denied

npkapur, 2016-09-21:
I pulled the ev3dev image from docker and installed some packages I want to have on the ev3.

I'm following the docs for creating a disk image from a docker image using `brickstrap`

I created an empty file `/brickstrap/_tar-exclude` inside the docker image to avoid issue #61.  However, now I'm getting the following error when running `./brickstrap/brickstrap.sh create-tar test test.tar`.

```
Checking docker image tar version...
tar 1.27.1-2+b1
Creating test.tar from test...
/bin/tar: /brickstrap/_tar-out/test.tar: Cannot open: Permission denied
/bin/tar: Error is not recoverable: exiting now
```

Any ideas?

dlech, 2016-09-21:
Which image did you pull? There are something like13 ev3dev images.

I'm thinking we should probably add `--user root` to all of the docker commands in `brickstrap.sh` to ensure that we are always running as root inside of the docker image.

dlech, 2016-09-21:
Also, `/brickstrap/_tar-out/` is bind-mounted to the current working directory on the host computer, so you need write permissions for the directory where where you run the `brickstrap` command.

npkapur, 2016-09-21:
I used this command: 

`docker pull ev3dev/debian-jessie-armel-cross`

I pulled the image this morning, so I'm guessing it is the most up to date.

dlech, 2016-09-21:
The `ev3dev/debian-jessie-armel-cross` image is for cross-compiling on a desktop system. It is not a good basis for an SD card image since it contains nearly 1GB of cross compiler programs that would be wasted space on the SD card. It is also missing a kernel and other essential packages needed to make a bootable image.

`ev3dev/ev3dev-jessie-ev3-generic` is the image we use to make official ev3dev releases for LEGO MINDSTORMS EV3 hardware. If you want a bare minimum image for EV3, then use `ev3dev/ev3dev-jessie-ev3-base` instead.

npkapur, 2016-09-23:
Our goal is to install some libraries in the image so we don't have to compile our binaries statically.  We'll give the `base` image a try.  Thanks!

closed 2016-10-13 by dlech