ev3dev/brickstrap #66: brickstrap error during create image from tar

FRozanski, 2018-09-27:
### System information:

- Ubuntu VERSION="18.04.1 LTS (Bionic Beaver)
- Docker version 17.05.0-ce, build 89658be

I am trying to do system image from docker image using brickstrap.
```
fifi@tesla-vb:~/DockerSavedImages$ sudo docker pull ev3dev/ev3dev-jessie-rpi2-generic
Using default tag: latest
latest: Pulling from ev3dev/ev3dev-jessie-rpi2-generic
c7a20c51e945: Pull complete 
06528a9c3e94: Pull complete 
8d4619ccef3f: Pull complete 
0dac42fa15e3: Pull complete 
a75873caa7f3: Pull complete 
4c64dabdfac7: Pull complete 
bcd2fcf342f2: Pull complete 
06583c7c51cd: Pull complete 
c0c14d947c13: Pull complete 
7238591e5f39: Pull complete 
00c727bf2c06: Pull complete 
Digest: sha256:525dd9f696c6e2fc446d491936886f5388495f73353b9fc33acea0e98bbbc601
Status: Downloaded newer image for ev3dev/ev3dev-jessie-rpi2-generic:latest
```
```
fifi@tesla-vb:~/DockerSavedImages$ sudo docker images
REPOSITORY                          TAG                 IMAGE ID            CREATED             SIZE
ev3dev/ev3dev-jessie-rpi2-generic   latest              92d10c50c4f1        6 months ago        596MB
```
```
fifi@tesla-vb:~/DockerSavedImages$ sudo docker save ev3dev/ev3dev-jessie-rpi2-generic > ev3dev.tar
fifi@tesla-vb:~/DockerSavedImages$ ls
ev3dev.tar
```
```
fifi@tesla-vb:~/DockerSavedImages$ sudo brickstrap create-image ev3dev.tar ev3dev.img
Creating ev3dev.img from ev3dev.tar...
libguestfs: error: tar_in: write error on directory: /:
fifi@tesla-vb:~/DockerSavedImages$ ls
ev3dev.img  ev3dev.tar
```
Then flashing ev3dev.img using Ether i getting empty folder.
Same result using brickstrap installed from apt-get and github.

Also installed this packages
```
fifi@tesla-vb:~/DockerSavedImages$ sudo apt-get install docker-engine libguestfs-tools qemu-user-static
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libguestfs-tools is already the newest version (1:1.36.13-1ubuntu3.1).
qemu-user-static is already the newest version (1:2.11+dfsg-1ubuntu7.5).
docker-engine is already the newest version (17.05.0~ce-0~debian-stretch).
The following package was automatically installed and is no longer required:
  pigz
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 22 not upgraded.
```
And perform this commands
```
fifi@tesla-vb:~/DockerSavedImages$ sudo update-guestfs-appliance
sudo: update-guestfs-appliance: command not found
fifi@tesla-vb:~/DockerSavedImages$ sudo usermod -a -G kvm $USER
fifi@tesla-vb:~/DockerSavedImages$ sudo chmod +r /boot/vmlinuz*
```

dlech, 2018-09-27:
You need to use `brickstrap create-tar` to create the tar file, not `docker save`.

FRozanski, 2018-09-27:
In this way?
```
fifi@tesla-vb:~/DockerSavedImages$ sudo brickstrap create-tar ev3dev/ev3dev-jessie-rpi2-generic ev3dev.tar
Checking docker image tar version...
tar 1.27.1-2+deb8u1
Creating ev3dev.tar from ev3dev/ev3dev-jessie-rpi2-generic...
done
Appending /brickstrap/_tar-only/*
done
Copying ev3dev.tar to /home/fifi/DockerSavedImages ...
done
fifi@tesla-vb:~/DockerSavedImages$ sudo brickstrap create-image ev3dev.tar ev3dev.img
Creating ev3dev.img from ev3dev.tar...
libguestfs: error: tar_in: write error on directory: /: 
fifi@tesla-vb:~/DockerSavedImages$ ls
ev3dev.img  ev3dev.tar
```
Ether still flash empty folder.

dlech, 2018-09-27:
Sounds like this bug: https://bugzilla.redhat.com/show_bug.cgi?id=1591617

This should be fixed in brickstrap already: https://github.com/ev3dev/brickstrap/commit/f7942f9f11481e19dbc6bb2c4b3d02a9dcb5a1a0

FRozanski, 2018-09-28:
Yeah, it was this bug. Now everything works fine. Thanks!
```
fifi@tesla-vb:~/DockerSavedImages$ sudo echo dash > /usr/lib/x86_64-linux-gnu/guestfs/supermin.d/zz-dash-packages
fifi@tesla-vb:~/DockerSavedImages$ sudo rm -rf /var/cache/.guestfs-$UID
fifi@tesla-vb:~/DockerSavedImages$ sudo brickstrap create-image ev3dev.tar ev3dev.img
Creating ev3dev.img from ev3dev.tar...
done
fifi@tesla-vb:~/DockerSavedImages$ ls
ev3dev.img  ev3dev.tar
```

dlech, 2018-09-28:
Did you install brickstrap from the ev3dev tool PPA? e.g. what is the output of `dpkg-query -s brickstrap | grep Version`?

FRozanski, 2018-09-28:
I installed brickstrap from this repo https://github.com/moriarty/ros-ev3/blob/master/brickstrap-build-status.md
```
fifi@tesla-vb:~$ dpkg-query -s brickstrap | grep Version
Version: 0.6.3
```

dlech, 2018-09-28:
Looks like we need to update the readme. The latest version is now hosted on launchpad:

    sudo add-apt-repository ppa:ev3dev/tools
    sudo apt update
    sudo apt upgrade

commit e1fdce55 (2018-09-28): Update README.md

New PPA and mention Ubuntu 18.04 libguestfs bug.

Closes #66
closed 2018-09-28 by dlech