ev3dev/brickstrap #68: Can't Boot Into Image

keenanjohnson, 2019-11-12:
I'm trying to get the image boot set up, but I can't seem to get in.

I'm using a Ubuntu VM on Parrellels.

parallels@parallels-Parallels-Virtual-Platform:~$ sudo docker run --rm -it ev3 su -l robot
standard_init_linux.go:211: exec user process caused "exec format error"

dlech, 2019-11-12:
you need to install `qemu-user-static` in Ubuntu.

keenanjohnson, 2019-11-12:
yes that worked! Thanks!

keenanjohnson, 2019-11-12:
Is it normal to get these QEMU errors?

parallels@parallels-Parallels-Virtual-Platform:~$ sudo docker run --rm -it ev3 su -l robot
[sudo] password for parallels: 
robot@0e39ee6f3cca:~$ sudo apt-get update
Unknown host QEMU_IFLA type: 47
Unknown host QEMU_IFLA type: 48
Unknown host QEMU_IFLA type: 43
Unknown host QEMU_IFLA type: 47
Unknown host QEMU_IFLA type: 48
Unknown host QEMU_IFLA type: 43

dlech, 2019-11-12:
you probably need to configure a DNS server in `/etc/resolve.conf`

closed 2020-08-24 by dlech