ev3dev/brickstrap #69: Can't Sudo inside Docker Containers

keenanjohnson, 2019-11-12:
I ran the setup to get the docker containers configured, but I can't seem to run sudo.

I get the following

no tty present and no askpass program specified docker

dlech, 2019-11-12:
You need to run docker with the `--tty` option.

keenanjohnson, 2019-11-12:
Hmmm running the command below didn't seem to resolve that. I still get the same no tty error.

Keenans-MBP:~ keenan$ docker run --tty --rm -it ev3dev/ev3dev-buster-ev3-generic su -l robot

dlech, 2019-11-12:
Try this to get a command prompt inside of the docker container: 

     docker run --tty --rm -it ev3dev/ev3dev-buster-ev3-generic bash

but be aware that when you exit, all changes are lost because of the `--rm` option.

dlech, 2019-11-12:
also, `--tty` is the same as `t` in `-it`

closed 2020-08-24 by dlech