ev3dev/brickstrap #73: 2023 Dec 8 Brickstrap successfully built bootable Buster & Bullseye .img, but now fails 2024 March

Growflavor, 2024-03-10:
Back on Dec 8 2023 I followed the clear & simple brickstrap instructions & quickly built myself bootable .img of the provided Buster beta & Bullseye beta.

I flashed these to 32GB SDcards with Balena Etcher & have been updating them 'onboard' (= the slow way ;-) with several of our EV3.  

Click the black arrow to show the key output details of the successful work on Dec 8 2023:

<details>
  <summary>Key Outputs from the successful Dec 8 2023 run -- Built using the apt installed Brickstrap inside a Docker/DockerDesktop integrated Windows 10 WSL2 Ubuntu 22.04:</summary>
  
Dec 8 2023 building of the default repository "Bullseye" beta .img :

  ```javascript
growflavor@AloE6700:~$ sudo brickstrap create-tar ev3dev/ev3dev-bullseye-ev3-generic ./Downloads/ev3dev11.tar			
Checking docker image tar version...			
WARNING: The requested image's platform (linux/arm) does not match the detected host platform (linux/amd64/v2) and no specific platform was requested			
tar 1.34			
Creating ./Downloads/ev3dev11.tar from ev3dev/ev3dev-bullseye-ev3-generic...			
WARNING: The requested image's platform (linux/arm) does not match the detected host platform (linux/amd64/v2) and no specific platform was requested			
done			
Appending /brickstrap/_tar-only/*			
done			
Copying ev3dev11.tar to /home/growflavor/Downloads ...			
Successfully copied 709MB to /home/growflavor/Downloads			
done			
growflavor@AloE6700:~$			
growflavor@AloE6700:~$ sudo brickstrap create-image ./Downloads/ev3dev11.tar ./Downloads/ev3dev.img			
Creating ./Downloads/ev3dev.img from ./Downloads/ev3dev11.tar...			
done			
growflavor@AloE6700:~$			
			
growflavor@AloE6700:~$ sudo brickstrap create-image ./Downloads/ev3dev11.tar ./Downloads/ev3dev11.img			
Creating ./Downloads/ev3dev11.img from ./Downloads/ev3dev11.tar...			
done			
growflavor@AloE6700:~$			
  ```

Dec 8 2023 building of the default repository "Buster" beta .img :
  
  ```javascript
growflavor@AloE6700:~$ sudo docker pull ev3dev/ev3dev-buster-ev3-generic		
[sudo] password for growflavor:		
Using default tag: latest		
latest: Pulling from ev3dev/ev3dev-buster-ev3-generic		
2660371b077d: Pull complete		
485a9f8c4061: Pull complete		
72a63efe0e62: Pull complete		
7b6a2a36d35f: Pull complete		
ea7a9ff66211: Pull complete		
4aed82a72658: Pull complete		
a4c4afafcc13: Pull complete		
d3ad7cedaa31: Pull complete		
c27837d2ae53: Pull complete		
Digest: sha256:3bd830fa9327f21e817fa353e8f188899a5eb2d1cdcb8ee9c0abe85c2ba52ee3		
Status: Downloaded newer image for ev3dev/ev3dev-buster-ev3-generic:latest		
docker.io/ev3dev/ev3dev-buster-ev3-generic:latest			
growflavor@AloE6700:~$ sudo docker tag ev3dev/ev3dev-buster-ev3-generic ev3d10		
growflavor@AloE6700:~$ sudo brickstrap create-tar ev3dev/ev3dev-buster-ev3-generic ./Downloads/ev3dev10.tar		
Checking docker image tar version...		
tar 1.30		
Creating ./Downloads/ev3dev10.tar from ev3dev/ev3dev-buster-ev3-generic...		
done		
Appending /brickstrap/_tar-only/*		
done		
Copying ev3dev10.tar to /home/growflavor/Downloads ...		
Successfully copied 882MB to /home/growflavor/Downloads		
done		
growflavor@AloE6700:~$ sudo brickstrap create-image ./Downloads/ev3dev10.tar ./Downloads/ev3dev10.img		
Creating ./Downloads/ev3dev10.img from ./Downloads/ev3dev10.tar...		
done		
growflavor@AloE6700:~$	
  ```

PuTTY Session output for the first login for the Buster image:

  ```javascript  
login as: robot			
Keyboard-interactive authentication prompts from server:			
| Password:			
End of keyboard-interactive prompts from server			
Linux ev3dev 4.19.85-ev3dev-3-beta2-ev3 #1 PREEMPT Sun Nov 24 16:49:49 CST 2019                                armv5tejl			
_____     _			
_____   _|___ /  __| | _____   __			
/ _ \ \ / / |_ \ / _` |/ _ \ \ / /			
|  __/\ V / ___) | (_| |  __/\ V /			
\___| \_/ |____/ \__,_|\___| \_/			
			
Debian buster on LEGO MINDSTORMS EV3!			
robot@ev3dev:~$ sudo apt update			
			
We trust you have received the usual lecture from the local System			
Administrator. It usually boils down to these three things:			
			
#1) Respect the privacy of others.			
#2) Think before you type.			
#3) With great power comes great responsibility.			
		
robot@ev3dev:~$ cat /etc/apt/sources.list			
deb http://httpredir.debian.org/debian buster main contrib non-free			
#deb-src http://httpredir.debian.org/debian buster main contrib non-free			
			
#deb http://security.debian.org/ buster/updates main contrib non-free			
#deb-src http://security.debian.org/ buster/updates main contrib non-free			
			
deb http://archive.ev3dev.org/debian buster main			
#deb-src http://archive.ev3dev.org/debian buster main			
robot@ev3dev:~$ sudo apt-get update			
[sudo] password for robot:			
Get:1 http://httpredir.debian.org/debian buster InRelease [122 kB]			
Hit:2 http://archive.ev3dev.org/debian buster InRelease			
Get:3 http://httpredir.debian.org/debian buster/non-free armel Packages [53.5 kB]			
Get:4 http://httpredir.debian.org/debian buster/main armel Packages [7648 kB]			
Get:5 http://httpredir.debian.org/debian buster/contrib armel Packages [37.6 kB]			
Fetched 7740 kB in 1min 19s (98.6 kB/s)			
Reading package lists... Done			
robot@ev3dev:~$			
robot@ev3dev:~$ python3 --version			
Python 3.7.3			
robot@ev3dev:~$			
  ```

PuTTY Session output for a recent login for the Dec 8 2023 Bullseye image:

  ```javascript  
login as: robot		
Keyboard-interactive authentication prompts from server:		
| Password:		
End of keyboard-interactive prompts from server		
Linux ev3dev 4.19.85-ev3dev-3-beta2-ev3 #1 PREEMPT Sun Nov 24 16:49:49 CST 2019 armv5tejl		
_____     _		
_____   _|___ /  __| | _____   __		
/ _ \ \ / / |_ \ / _` |/ _ \ \ / /		
|  __/\ V / ___) | (_| |  __/\ V /		
\___| \_/ |____/ \__,_|\___| \_/		
		
Debian bullseye on LEGO MINDSTORMS EV3!		
Last login: Sat Mar  9 20:05:22 2024 from 192.168.1.109		
robot@ev3dev:~$ cat /etc/apt/sources.list		
deb http://httpredir.debian.org/debian bullseye main contrib non-free		
#deb-src http://httpredir.debian.org/debian bullseye main contrib non-free		
		
#deb http://security.debian.org/ bullseye/updates main contrib non-free		
#deb-src http://security.debian.org/ bullseye/updates main contrib non-free		
		
deb http://archive.ev3dev.org/debian bullseye main		
#deb-src http://archive.ev3dev.org/debian bullseye main		
robot@ev3dev:~$ sudo apt update		
		
We trust you have received the usual lecture from the local System		
Administrator. It usually boils down to these three things:		
		
#1) Respect the privacy of others.		
#2) Think before you type.		
#3) With great power comes great responsibility.		
		
[sudo] password for robot:		
Get:1 http://httpredir.debian.org/debian bullseye InRelease [116 kB]		
Get:2 http://archive.ev3dev.org/debian bullseye InRelease [4892 B]		
Get:3 http://httpredir.debian.org/debian bullseye/contrib armel Packages [37.3 kB]		
Get:4 http://httpredir.debian.org/debian bullseye/non-free armel Packages [51.6 kB]		
Get:5 http://httpredir.debian.org/debian bullseye/main armel Packages [7781 kB]		
Fetched 7991 kB in 5min 57s (22.4 kB/s)		
Reading package lists... Done		
Building dependency tree... Done		
Reading state information... Done		
153 packages can be upgraded. Run 'apt list --upgradable' to see them.		
robot@ev3dev:~$		
  ```
</details>
	
However, now in 2024 March,  when I run brickstrap in the same setup, libguestfs fails with an error & Etcher reports that the .img has 'no filesystem'...it seems that now for some reason libguestfs can not find a "suitable kernal"

`growflavor@AloE6700:~$ sudo brickstrap create-image ./Downloads/ev3dev11march2024Test.tar ./Downloads/ev3dev11march2024Test.img
Creating ./Downloads/ev3dev11march2024Test.img from ./Downloads/ev3dev11march2024Test.tar...
libguestfs: error: /usr/bin/supermin exited with error status 1.
To see full error messages you may need to enable debugging.
Do:
  export LIBGUESTFS_DEBUG=1 LIBGUESTFS_TRACE=1
and run the command again.  For further information, read:
  http://libguestfs.org/guestfs-faq.1.html#debugging-libguestfs
You can also run 'libguestfs-test-tool' and post the *complete* output
into a bug report or message to the libguestfs mailing list.
growflavor@AloE6700:~$`

I would greatly appreciate your insights on what to try next or if there is some simple change I need to add to brickstrap.sh, etc... 

Thank you!

dlech, 2024-03-10:
The first thing I would try is doing what the error message says and set the environment variables to get more details about the error.

Growflavor, 2024-03-11:
> The first thing I would try is doing what the error message says and set the environment variables to get more details about the error.

I would be grateful if you would clarify from where brickstrap needs to get the appropriate kernel to build the proper EV3 boot partition for the brickstrap created .img

I have been thinking that brickstrap/libguestfs-tools would pull the kernel from the custom brickstrap .tar built from the docker image...because I have assumed that the Ubuntu x64 OS would not have anywhere the appropriate EV3 armel kernel.  Is this correct?  Or do I need to copy the kernel from the docker EV3 container to the Ubuntu host "/boot/vmlinuz"?

Since all I see in the trace are:
`supermin: build: visiting /usr/lib/x86_64-linux-gnu...

supermin: kernel: kernel version of /boot/vmlinuz-5.19.0-1022-oracle = 5.19.0-1022-oracle (from content)
supermin: kernel: picked modules path /lib/modules/5.19.0-1022-oracle
supermin: failed to find a suitable kernel (host_cpu=x86_64).`

As far as I can tell from the debug trace, libguestfs-tools are not looking in the correct place(s) for the armel kernel...or perhaps you see one or more other obvious issues...

i.e. where should libguestfs-tools look for the kernel & if it is indeed in the .tar then what should I make sure to have set so that it can actually look there?    

Also, I just an not clear what could have changed (something internal to Ubuntu, or windows, or docker desktop, or...) since this happens now on all my computers (win 10 & win 11) (even if I restore the WSL2 backup image I took in Dec within a few days of it making the successful EV3 .img files) where as brickstrap was successful on these machines back in Dec...:

<details>
  <summary>libguestfs: trace</summary>

_________________________________________________________
  ```javascript  
growflavor@AloEnergia:~$ libguestfs-test-tool
     ************************************************************
     *                    IMPORTANT NOTICE
     *
     * When reporting bugs, include the COMPLETE, UNEDITED
     * output below in your bug report.
     *
     ************************************************************
libguestfs: trace: set_verbose true
libguestfs: trace: set_verbose = 0
libguestfs: trace: set_verbose true
libguestfs: trace: set_verbose = 0
LIBGUESTFS_TRACE=1
LIBGUESTFS_DEBUG=1
PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/usr/lib/wsl/lib:/mnt/c/Program Files/Microsoft/jdk-11.0.16.101-hotspot/bin:/mnt/c/Program Files/Common Files/Oracle/Java/javapath:/mnt/c/Program Files (x86)/Common Files/Oracle/Java/javapath:/mnt/c/Windows/system32:/mnt/c/Windows:/mnt/c/Windows/System32/Wbem:/mnt/c/Windows/System32/WindowsPowerShell/v1.0/:/mnt/c/Windows/System32/OpenSSH/:/mnt/c/Program Files/dotnet/:/mnt/c/Program Files (x86)/NVIDIA Corporation/PhysX/Common:/mnt/c/Program Files/Microsoft VS Code/bin:/mnt/c/Program Files/MATLAB/MATLAB Runtime/v912/runtime/win64:/mnt/c/Program Files (x86)/QuickTime/QTSystem/:/mnt/c/Program Files (x86)/Common Files/Acronis/SnapAPI/:/mnt/c/Program Files (x86)/Common Files/Acronis/VirtualFile/:/mnt/c/Program Files (x86)/Common Files/Acronis/VirtualFile64/:/mnt/c/Program Files (x86)/Common Files/Acronis/FileProtector/:/mnt/c/Program Files (x86)/Common Files/Acronis/FileProtector64/:/mnt/c/Program Files (x86)/IVI Foundation/VISA/WinNT/Bin:/mnt/c/Program Files/Git/cmd:/mnt/c/Program Files/PuTTY/:/mnt/c/Program Files/Docker/Docker/resources/bin:/mnt/c/Program Files/PowerShell/7/:/mnt/c/Users/growf/AppData/Local/Microsoft/WindowsApps:/snap/bin
XDG_RUNTIME_DIR=/run/user/1000/
SELinux: sh: 1: getenforce: not found
libguestfs: trace: add_drive_scratch 104857600
libguestfs: trace: get_tmpdir
libguestfs: trace: get_tmpdir = "/tmp"
libguestfs: trace: disk_create "/tmp/libguestfsqkr6xa/scratch1.img" "raw" 104857600
libguestfs: trace: disk_create = 0
libguestfs: trace: add_drive "/tmp/libguestfsqkr6xa/scratch1.img" "format:raw" "cachemode:unsafe"
libguestfs: trace: add_drive = 0
libguestfs: trace: add_drive_scratch = 0
libguestfs: trace: get_append
libguestfs: trace: get_append = "NULL"
guestfs_get_append: (null)
libguestfs: trace: get_autosync
libguestfs: trace: get_autosync = 1
guestfs_get_autosync: 1
libguestfs: trace: get_backend
libguestfs: trace: get_backend = "direct"
guestfs_get_backend: direct
libguestfs: trace: get_backend_settings
libguestfs: trace: get_backend_settings = []
guestfs_get_backend_settings: []
libguestfs: trace: get_cachedir
libguestfs: trace: get_cachedir = "/var/tmp"
guestfs_get_cachedir: /var/tmp
libguestfs: trace: get_hv
libguestfs: trace: get_hv = "/usr/bin/qemu-system-x86_64"
guestfs_get_hv: /usr/bin/qemu-system-x86_64
libguestfs: trace: get_memsize
libguestfs: trace: get_memsize = 1280
guestfs_get_memsize: 1280
[... 130 lines cut ...]
  Candidate: 5:25.0.4-1~ubuntu.22.04~jammy
  Version table:
 *** 5:25.0.4-1~ubuntu.22.04~jammy 500
        500 https://download.docker.com/linux/ubuntu jammy/stable amd64 Packages
        100 /var/lib/dpkg/status
     5:25.0.3-1~ubuntu.22.04~jammy 500
        500 https://download.docker.com/linux/ubuntu jammy/stable amd64 Packages
     5:25.0.2-1~ubuntu.22.04~jammy 500
    ...
growflavor@AloEnergia:~$ kvm-ok
INFO: /dev/kvm exists
KVM acceleration can be used
growflavor@AloEnergia:~$ apt-cache policy brickstrap
brickstrap:
  Installed: 0.6.8
  Candidate: 0.6.8
  Version table:
 *** 0.6.8 500
        500 https://ppa.launchpadcontent.net/ev3dev/tools/ubuntu jammy/main amd64 Packages
        100 /var/lib/dpkg/status
growflavor@AloEnergia:~$
  ```  
</details>

Thanks for your further insights!

dlech, 2024-03-11:
> would be grateful if you would clarify from where brickstrap needs to get the appropriate kernel

The kernel comes from a debian package just like everything else. So for any ev3dev-specific packages like that one, we need to update the package for the newer versions of debian.

I also recall that on Ubuntu, the permissions of the host kernel in `/boot` needs to have the permissions fixed to be compatible with libguestfs. This fix has to be done every time a new kernel is installed on the host computer.

Growflavor, 2024-03-19:
apparently some kernels (such as vmlinuz-5.19.0-1022-oracle ) are not suitable...[discussion at this link suggests a solution](https://patchwork.ozlabs.org/project/uboot/patch/20210621185156.9108-2-alpernebiyasak@gmail.com/):

sudo apt-get install -y linux-image-kvm

Only set /boot/vmlinu* as readable

-rw-r--r-- 1 root root  6960424 Feb  8 15:04 vmlinuz-5.15.0-1051-kvm

"As this kernel would be only used for virtualization, we can use the kernel package specialized for that. On Ubuntu systems kernel images are not readable by non-root users, so explicitly add read permissions with chmod as well."

[image]
[image]

I will check if this also works on my Windows 11 WSL2 setup.

Growflavor, 2024-03-19:
verified as well on Windows 11 WSL2 Ubuntu 22.04...Brickman successfully completes tasks again once a different kernel such as the linux-image-kvm is installed and made readable...

sudo apt-get install -y linux-image-kvm
sudo chmod +r /boot/vmlinu*

[image]

Growflavor, 2024-05-19:
I have been regularly loosing Brickstrap functionality with the  LIBGUESTFS error...apparently Ubuntu has an unattended periodic automated apt security updates.

So for my 'dedicated working' Brickstrap WSL2 Ubuntu image I've decided to disable that feature by setting both the following values to 0:

sudo nano /etc/apt/apt.conf.d/20auto-upgrades

[image]