ev3dev/brickstrap #74: macos support

Muresan73, 2024-12-04:
Have someone tried to make the scripts work with macos (aarch64) ?

My understanding is that the script just create an artifact by extracting the necessary system files from the docker image.
I try to create an image from the `ev3dev/ev3dev-bullseye-ev3-generic` image from dockerhub,
but i am getting
```
Cannot read /boot/vmlinuz-* needed by guestfish
Set permission with 'sudo chmod +r /boot/vmlinuz-*'
```
It seems to me that it needs some files from my host system.

Can someone clarify what i am doing wrong ?
It would make my life easier if i dont need an extra debian vm if I am on the go  :)

dlech, 2024-12-04:
It requires a Linux host. If you have docker already, you could use that instead of a full VM.

Muresan73, 2024-12-04:
So should I use docker in docker or is it possible to run brickstrap in the target container?

It is anyhow better since I was struggling to hook up the qemu docker extension on ubuntu anyway :/

dlech, 2024-12-04:
Hmm... not sure how running docker in docker would work. I guess you would just have to try it.

You shouldn't need any special docker extensions on Ubuntu. Just `apt install qemu-user-static`.

Muresan73, 2024-12-04:
> It requires a Linux host. If you have docker already, you could use that instead of a full VM.

then i am confused what you mean by linux host in this context
it still need docker on that system

i've installed qemu-user-static but its still not working. My guess is that aarch host is not supported just x86 ( i am using aarch ubuntu server 24 lts )

i will come back to you if i figured it out

Muresan73, 2024-12-05:
Docker Outside of Docker (dood) was the solution, using the system docker socket within the docker container
```bash
docker run -it -v /var/run/docker.sock:/var/run/docker.sock docker:latest sh
```

And from there the system docker images can be accessed and run `brickstrap` on them. 

---
*ps: For those who might come across this thread in the future:*
This image is based on a minimal alpine so installing bash with `apk install bash` can be necessary.
If you are done with setting up the container you can just create an image out of it for later use with 
```bash
docker ps # get the container-id
docker commit <container-id> <new-image-name>:<tag>
```

closed 2024-12-05 by Muresan73